hubsif/arduino-dali #2: dali setup short address example

CurseStaff, 2022-10-03:
HI hubsif,

thank you for this awesome work,

can you share an example of driver setup code please ?

thank !

hubsif, 2022-10-04:
Hi @CurseStaff,

What exactly do you mean by "an example of driver setup code"? You mean how to set the addresses of the drivers (in DALI terms: commission ballasts?)

hubsif, 2022-10-05:
Hi @CurseStaff,

Originally, I wanted to write more documentation and also provide a sample, but I didn't get to that yet.

Please find a minimal example with comments below. Please also have a look at the documentation [here](https://hubsif.github.io/arduino-dali/class_dali_class.html#acd38aec28e257d510f152fb50cca3848).

```c++
// DALI rx and tx pins
const byte DALI_TX_PIN = 1;
const byte DALI_RX_PIN = 0;

void setup() {
  Dali.begin(DALI_TX_PIN, DALI_RX_PIN); // start DALI
}

void loop() {
  // DALI tick (only required during commissioning)
  Dali.commission_tick();
}

//
// functions that can be called to start different kinds of commissioning
//
void commission_bus() {
  // (Re-)Commission all ballasts on the bus. overwrites existing addressing

  // Note that in DALI commissioning ballasts are selected randomly each time.
  // So, if the bus had been commissioned before, addresses can be different afterwards
  Dali.commission();
}

void commission_new() {
  // Commission ballasts on the bus that don't have an address yet. Assign addresses starting from 'startAddress' upwards.

  // Note that in this example if there were two new (addressless) ballasts and an existing ballast
  // with address 6 on the bus, you'd end up with two ballasts having the same address 6.
  // Ensuring that there is only one new ballast on the bus this can be used to e.g. replace
  // a defective ballast with address 5
  byte startAddress = 5;
  bool onlyNew = true
  Dali.commission(startAddress, true);
}
```

I just realized the issue with the double addressing. Perhaps I'll fix that in an upcoming version.

laihappy, 2022-10-05:
Good Job! does your program support dali 2.0?

closed 2023-10-02 by hubsif