hubsif/arduino-dali #6: also receive commands

thewhobox, 2023-06-29:
Is it possible to enhance this lib to receive commands from the Dali bus?
With this we could build a dali monitor or even own dali devices.

I think a good startpoint could be:
https://github.com/hubsif/arduino-dali/blob/28aafd04348bb17abc3dc1fb5b110b98bf475cee/src/DaliBus.cpp#L208

If we dont expect a response it is a command.

Bill-Gaytes, 2023-10-21:
Hi 'thewhobox',
I also want to build my own DALI devices (DALI Slave). Did you get any further with this idea ?

thewhobox, 2023-10-21:
@Bill-Gaytes yes I got it working. See pending pull request.

Bill-Gaytes, 2023-10-22:
Hi Mike,
Firstly thanks you for your reply. I will admit that I don't have a great
deal of experience with the workings of GitHub so that's why I didn't look
in Pull requests.
I can see the Pull request where you made changes to Dali.cpp and then 3
Commits to make changes to Dali.h, DaliBus.cpp and DaliBus.h ?
Would I be correct that I now need to take your changes and update the Dali
library ?

I have all the Dali hardware for this but not really certain with the
software. The example (Dali Blink) is not really that helpful either.
Sorry for my naivety, but we all have keep learning.

Cheers Bill

thewhobox, 2023-10-22:
Hey, there are two options to get it work in your project.
- you can make the changes manually by copy the changes in DaliBus.cpp and DaliBus.h and then set 
```C++
DaliBus.receivedCallback([](uint8_t *data, uint8_t len) -> void {
    //do stuff with the raw message here
});
```
- use my fork branch: https://github.com/thewhobox/arduino-dali/tree/feature-add-receiving

but you will have to parse these messages by yourself.

Bill-Gaytes, 2023-10-24:
Hi Mike,

Many thanks, I followed the branch and updated the library, looks to be OK.

Other than the dali_blink sketch, do you know of any other examples using
this library that I can have a look at so I can learn a bit more ?

Cheers

thewhobox, 2023-10-24:
Depends on what you want to learn.
If you want to build your own Dali Slave i highly recommend you to learn about the commands and what are they doing:
https://onlinedocs.microchip.com/pr/GUID-0CDBB4BA-5972-4F58-98B2-3F0408F3E10B-en-US-1/index.html?GUID-DA5EBBA5-6A56-4135-AF78-FB1F780EF475

You would also need to implement memory banks, if you want your device to be setup by Dali Cockpit or something else:
https://infosys.beckhoff.com/english.php?content=../content/1033/tcplclib_tc3_dali/6940982539.html&id=

hubsif, 2023-10-25:
Hi @Bill-Gaytes!
You might also want to have a look at my project [DALI-MQTT-actuator](https://github.com/hubsif/DALI-MQTT-actuator) which is using this library.

closed 2024-09-16 by thewhobox