hubsif/arduino-dali #8: Dali Bus Commissioning not working

mightChamp, 2023-07-28:
There are 3 Dali Blast devices in Bus, which can be operated using broadcast command like Direct Arc Power Control/Up/Down, etc..

but when using below function to Commission the Dali bus, then commissioning not happens, So can't use short address of individual Blast Device.

```
void commission_bus() {
  // (Re-)Commission all ballasts on the bus. overwrites existing addressing

  // Note that in DALI commissioning ballasts are selected randomly each time.
  // So, if the bus had been commissioned before, addresses can be different afterwards
  Dali.commission();
  while(1) {
    Dali.commission_tick();
    delay(50);
    Serial.print("commissionState = ");
    Serial.println(Dali.commissionState);
    if(Dali.commissionState == Dali.COMMISSION_OFF)
    {
      break;
    }
  }
}
```

Below is log of commissioning States:

```
[11:13:42.474] commissionState = 2
[11:13:42.524] commissionState = 3
[11:13:42.575] commissionState = 4
[11:13:42.624] commissionState = 5
[11:13:42.674] commissionState = 6
[11:13:42.724] commissionState = 7
[11:13:42.775] commissionState = 8
[11:13:42.824] commissionState = 8
[11:13:42.875] commissionState = 8
[11:13:42.925] commissionState = 9
[11:13:42.975] commissionState = 11
[11:13:43.025] commissionState = 12
[11:13:43.075] commissionState = 13
[11:13:43.125] commissionState = 14
[11:13:43.174] commissionState = 10
[11:13:43.225] commissionState = 11
[11:13:43.275] commissionState = 12
[11:13:43.325] commissionState = 13
[11:13:43.375] commissionState = 14
[11:13:43.425] commissionState = 10
[11:13:43.475] commissionState = 11
[11:13:43.525] commissionState = 12
[11:13:43.575] commissionState = 13
[11:13:43.625] commissionState = 14
[11:13:43.675] commissionState = 10
[11:13:43.725] commissionState = 11
[11:13:43.775] commissionState = 12
[11:13:43.825] commissionState = 13
[11:13:43.875] commissionState = 14
[11:13:43.925] commissionState = 10
[11:13:43.975] commissionState = 11
[11:13:44.025] commissionState = 12
[11:13:44.075] commissionState = 13
[11:13:44.125] commissionState = 14
[11:13:44.174] commissionState = 10
[11:13:44.224] commissionState = 11
[11:13:44.274] commissionState = 12
[11:13:44.325] commissionState = 13
[11:13:44.375] commissionState = 14
[11:13:44.425] commissionState = 10
[11:13:44.475] commissionState = 11
[11:13:44.525] commissionState = 12
[11:13:44.575] commissionState = 13
[11:13:44.625] commissionState = 14
[11:13:44.675] commissionState = 10
[11:13:44.725] commissionState = 11
[11:13:44.775] commissionState = 12
[11:13:44.825] commissionState = 13
[11:13:44.875] commissionState = 14
[11:13:44.925] commissionState = 10
[11:13:44.975] commissionState = 11
[11:13:45.025] commissionState = 12
[11:13:45.076] commissionState = 13
[11:13:45.124] commissionState = 14
[11:13:45.175] commissionState = 10
[11:13:45.225] commissionState = 11
[11:13:45.275] commissionState = 12
[11:13:45.326] commissionState = 13
[11:13:45.375] commissionState = 14
[11:13:45.425] commissionState = 10
[11:13:45.475] commissionState = 11
[11:13:45.525] commissionState = 12
[11:13:45.575] commissionState = 13
[11:13:45.624] commissionState = 14
[11:13:45.675] commissionState = 10
[11:13:45.725] commissionState = 11
[11:13:45.776] commissionState = 12
[11:13:45.824] commissionState = 13
[11:13:45.875] commissionState = 14
[11:13:45.925] commissionState = 10
[11:13:45.975] commissionState = 11
[11:13:46.025] commissionState = 12
[11:13:46.075] commissionState = 13
[11:13:46.125] commissionState = 14
[11:13:46.175] commissionState = 10
[11:13:46.225] commissionState = 11
[11:13:46.275] commissionState = 12
[11:13:46.325] commissionState = 13
[11:13:46.374] commissionState = 14
[11:13:46.425] commissionState = 10
[11:13:46.475] commissionState = 11
[11:13:46.525] commissionState = 12
[11:13:46.575] commissionState = 13
[11:13:46.625] commissionState = 14
[11:13:46.675] commissionState = 10
[11:13:46.725] commissionState = 11
[11:13:46.775] commissionState = 12
[11:13:46.825] commissionState = 13
[11:13:46.875] commissionState = 14
[11:13:46.925] commissionState = 10
[11:13:46.975] commissionState = 11
[11:13:47.025] commissionState = 12
[11:13:47.077] commissionState = 13
[11:13:47.125] commissionState = 14
[11:13:47.175] commissionState = 10
[11:13:47.225] commissionState = 11
[11:13:47.275] commissionState = 12
[11:13:47.325] commissionState = 13
[11:13:47.375] commissionState = 14
[11:13:47.425] commissionState = 10
[11:13:47.475] commissionState = 11
[11:13:47.525] commissionState = 12
[11:13:47.575] commissionState = 13
[11:13:47.625] commissionState = 14
[11:13:47.675] commissionState = 10
[11:13:47.725] commissionState = 11
[11:13:47.775] commissionState = 12
[11:13:47.825] commissionState = 13
[11:13:47.875] commissionState = 14
[11:13:47.925] commissionState = 10
[11:13:47.975] commissionState = 11
[11:13:48.025] commissionState = 12
[11:13:48.079] commissionState = 13
[11:13:48.124] commissionState = 14
[11:13:48.175] commissionState = 10
[11:13:48.225] commissionState = 11
[11:13:48.275] commissionState = 12
[11:13:48.325] commissionState = 13
[11:13:48.375] commissionState = 14
[11:13:48.424] commissionState = 10
[11:13:48.475] commissionState = 11
[11:13:48.524] commissionState = 12
[11:13:48.575] commissionState = 13
[11:13:48.625] commissionState = 14
[11:13:48.675] commissionState = 10
[11:13:48.725] commissionState = 11
[11:13:48.775] commissionState = 12
[11:13:48.825] commissionState = 13
[11:13:48.875] commissionState = 14
[11:13:48.925] commissionState = 10
[11:13:48.975] commissionState = 11
[11:13:49.025] commissionState = 12
[11:13:49.079] commissionState = 13
[11:13:49.125] commissionState = 14
[11:13:49.175] commissionState = 15
[11:13:49.225] commissionState = 16
[11:13:49.275] commissionState = 17
[11:13:49.325] commissionState = 21
[11:13:49.380] commissionState = 0
```

milo1000, 2023-07-28:
For me it helped to increase DALI_TE_MAX limit, like:
`const unsigned long DALI_TE_MAX = (130 * DALI_TE) / 100;`
I am not sure it's the same problem though.

hubsif, 2023-07-28:
Hi @mightChamp!
From the log I can see that the first DALI ballast is found, but the process fails in verifying the programmed short address, code [here](https://github.com/hubsif/arduino-dali/blob/28aafd04348bb17abc3dc1fb5b110b98bf475cee/src/Dali.cpp#L214):

```c++
      case COMMISSION_VERIFYSHORTRESPONSE:
        if (DaliBus.getLastResponse() == 0xFF) {  // <<<= not true
          nextShortAddress++;
          commissionState = COMMISSION_WITHDRAW;
```
and thus commissioning is stopped.

I'm not sure what's causing this, though. I've seen ballasts that are somewhat slow (e.g. some OSRAM ballasts seem to take longer to generate a random address, that's why the code [waits a bit here](https://github.com/hubsif/arduino-dali/blob/28aafd04348bb17abc3dc1fb5b110b98bf475cee/src/Dali.cpp#L158)).
Perhaps your ballast takes some extra time to set its short address 🤷‍♂️ . It might help to log the result of `DaliBus.getLastResponse()`.

Also, you could also try setting your `delay(50);` to a higher value (e.g. 100) and see if that makes a change already.

What brand and type are your ballasts? Are they all the same? If you always had all three on the bus, try with only one at a time.

hubsif, 2023-07-28:
> For me it helped to increase DALI_TE_MAX limit, like:

@milo1000 
Although I don't think this is the issue here, you newer shared that in [your issue](https://github.com/hubsif/arduino-dali/issues/3). Perhaps it doesn't hurt to generally increase the tolerance, I'll check.

mightChamp, 2023-07-28:
Hello @hubsif  and @milo1000 ,

Thanks for your response. We are using this library on ESP32 Development Board, and now the issue is solved by using a 4.7k pullup resistor on Dali Tx and Rx Pins of ESP32.

Thanks once again.

closed 2023-07-28 by mightChamp