hubsif/arduino-dali #9: How commissioning process takes place in DALI system?

mightChamp, 2023-08-01:
I finding it difficult to understand some of the points
I understand Steps Which follow in commission_tick() Function but unable to understand following points.

1) in only 24 itterations how its identify right dali devices?
2) if two Dali Device have same Address in single bus then how to change address without remove device from dali bus.

thewhobox, 2023-10-02:
1. in commission mode all devices receive and process commands when there own random address (3 bytes) is equal or smaller than the SEARCHADDR High/Middle/Low. if one ore more devices succeed this condition they will answer to a query request.
It now searchs for addresses in the lower half, if there is an address ist searches in the lower half of the half. if there is no address it searches in the upper half:
|- - - -|
|- -|- -|
|x|-|-|-|
there are youtube videso which explain the algorythmus on a real bus search

2. its like commissioning but with less steps
 - initialize all devices
 - randomize
 - search all devices by its long address (like if they were new no the bus), after you found a device request its short address
 - you now have the short address which is used twice with the long address (ex. A2: 0xABCDEF and A2: 0x123456
 - search for the long address - 1 which you want to change the short address (ex. 0x123455)
 - withdraw all other devices so they wont respond to the commands
 - search for the long address (ex. 0x123456)
 - assign a new short address
 - termintate

hubsif, 2023-10-02:
Thank you, @thewhobox, for providing the correct answers! I had missed the original request.
Adding to 1.: very simplified you could say it checks for all 24 bits of the long address = 24 iterations.

Assuming this can be closed then.

closed 2023-10-02 by hubsif