hubsif/arduino-dali #10: Unable to communicate over the DALI Bus

jonathanlundstrom, 2023-10-12:
Hi,

I'm attempting to use this library with an Arduino Uno and a [DALI 2 CLICK Module](https://www.mikroe.com/dali-2-click) but I'm having very limited success. The Dali bus is parallell powered with a 12V power supply an the DALI bus on the Click module, and all seems in order on that side. The TX pin on the module is connected to `digital pin 2`, and RX is connected to `digital pin 3` on the Arduino, but I've also attempted to use `A0` as the input pin. It's not clear to me if the RX signal is digital or not?

Either way, this is how my code looks:
```cpp
#include <Arduino.h>
#include <Dali.h>

const int DALI_TX_PIN = 2;
const int DALI_RX_PIN = 3;

void setup() {
  Serial.begin(9600);
  Dali.begin(DALI_TX_PIN, DALI_RX_PIN, false);
  Dali.commission(0, false);
}

void loop() {
  while (Dali.commissionState != DaliClass::commissionStateEnum::COMMISSION_OFF) {
    Serial.println("Current commission state: " + (String) Dali.commissionState);
    Serial.println("Next short address: " + (String) Dali.nextShortAddress);
    Dali.commission_tick();
    delay(500);
  }

  Serial.println("Current state: " + (String) Dali.commissionState);
  Serial.println("Number of ballasts found: " + (String) Dali.nextShortAddress);
  
  // Turn off ballast on short code 1
  Dali.sendArc(0, Dali.CMD_RECALL_MAX);
  delay(1000);

  // Turn off ballast at short code 1
  Dali.sendArc(0, Dali.CMD_OFF);
  delay(1000);
}
```

As you can see, I'm attempting to commission the devices in order to assign short codes so that I can later use them to simply blink the lamps connected to one of the DALI LED-drivers on the bus. This is the output from my code:

```sh
Current commission state: 1
Next short address: 0
Current commission state: 2
Next short address: 0
Current commission state: 3
Next short address: 0
Current commission state: 4
Next short address: 0
Current commission state: 5
Next short address: 0
Current commission state: 6
Next short address: 0
Current commission state: 7
Next short address: 0
Current commission state: 8
Next short address: 0
Current commission state: 9
Next short address: 0
Current commission state: 11
Next short address: 0
Current commission state: 12
Next short address: 0
Current commission state: 13
Next short address: 0
Current commission state: 14
Next short address: 0
Current commission state: 21
Next short address: 0
```

I've also had to set `active_low` to false on the `Dali.begin` method, because otherwise I get stuck on commission state 2 (INIT2). Doing so means I get all the way to state 14 before the code terminates, supposedly because it doesn't find any ballasts on the bus.

Is there something obvious that I am missing? 😭

Thank you!

jonathanlundstrom, 2023-10-12:
> Hi @jonathanlundstrom <https://github.com/jonathanlundstrom>!
>
> Is there something obvious that I am missing? 😭
>
> To me there is 😄. Without having checked all the info you provided on
> your physical setup, I think this: delay(500); in your while loop might
> be the issue, as it blocks the commissioning process for 500ms while there
> are shorter timeouts on the bus that must be adhered to.
> So, please remove this line and in order to still be able to debug but not
> burst your serial line, use a non-blocking delay for your serial output.
> You can find many samples in the internet, e.g. here
> <https://fvdm.com/code/blocking-vs-non-blocking-timers-in-arduino-and-particle>
> If it still doesn't work feel free to share your new results.

Hi,

Thank you for your reply! I've also contemplated this and have since
removed them, but unfortunately it doesn't make any difference. I'm using
an Arduino Uno but I'm assuming that processor should be more than capable
to handle this kind of communication?

I've also switched out 12V power supply in favor of a lab PSU set to 24 and
clamped to 250mAh.

Still, after step 14 it will continuously fail and move on to 21, but I
have also noticed that the 250mAh clamping doesn't seem to be sufficient as
the hardware will attempt to draw more power, causing the PSU to restart.

Since there are no schematics available on the repository, may I please ask
about your test setup?

Regards,
Jonathan

hubsif, 2023-10-12:
> after step 14 it will continuously fail and move on to 21

If not clear to you already, this means that no ballast (LED driver) was found.

> Since there are no schematics available on the repository, may I please ask
about your test setup?

My test setup is a custom board driven by an Arduino Mini Pro as in [this project](https://github.com/hubsif/DALI-MQTT-actuator).

> the hardware will attempt to draw more power, causing the PSU to restart

The DALI bus communication works by pulling the bus voltage (9.5-22.5V) to ground, but this doesn't last any longer than ~1ms. Are you certain you have set `active_low` correctly? (from what I understand from their schematics, the red LED of the CLICK module needs to be on in idle mode)

To be able to further assist you with your physical setup I would need you to share your schematics. Sometimes it can be as simple as not having GND connected 😉.

hubsif, 2023-10-12:
> Are you certain you have set active_low correctly?

I've had a closer look at the schematics of the CLICK module and if you connected the Arduino pins directly to the module, `active_low` needs to stay default=true.

jonathanlundstrom, 2023-10-12:
I think our problem is the power supply. We're using a Lab PSU set to 16V (also tried 24V) clamped to 250mAh but we're thinking that it's probably resetting from a short too slowly. Most likely we'll need to use a PSU designed for use with Dali – or what are your thoughts on this?

We've also been experimenting with [this library](https://github.com/qqqlab/DALI-Lighting-Interface) but with the same exact results unfortunately.

jonathanlundstrom, 2023-10-12:
Also, this is a rough draft of my schematic. It's late here so please excuse that it looks insane 😂

[image]

hubsif, 2023-10-12:
> or what are your thoughts on this?

I don't think the power supply is the issue here. From this:

> I get stuck on commission state 2 (INIT2)

I suspect your timer to not work, which leads to the bus getting pulled down permanently and your PS eventually shutting down. Perhaps you can confirm by checking the temperature of the fuse on the CLICK module when that happens. It should not get hot.
Perhaps I can help you look into this timer issue tomorrow, like you said, it's late already 😉.

Also, please adjust your power supply to 16V, the default voltage for DALI. Your 24V are out of the DALI specs (i.e. 16V±6.5V).

hubsif, 2023-10-12:
for starters, set your `active_low` back to true and disconnect the DALI bus from the module. Then check if the commissioning process still gets stuck at INIT2 or if it runs through without finding a ballast.

jonathanlundstrom, 2023-10-18:
The problem was the power supply. After getting a DALI compliant PSU at 16V 250mA things started working as expected using [this library](https://github.com/qqqlab/DALI-Lighting-Interface). I suspect this one would work equally well. Thank you for your support! 👍

closed 2023-10-18 by jonathanlundstrom