hubsif/arduino-dali #11: Sending color command with Arduino according to IEC 62386-209

anilaktas, 2024-05-27:
The library works fine, but how can I send color commands according to IEC 62386-209 with Arduino?  The example command is as follows: Address 0, r=0%, g=100%, b=keep, w=100% with RGBW cutoff Dali.sendArc(0,"00FEFFFE");

hubsif, 2024-05-30:
I've never done it myself, but perhaps [this explanation here](https://electronics.stackexchange.com/a/605066) helps understand the general idea of 209.

thewhobox, 2024-09-13:
You are very limited with 2 bytes of commands.
So you cant fit rgb in one command.
What dali does:
 - store every value in a dtr
 - tell the device to use this as rgb
 - activate it

This is what i do for rgb:
```C++
        sendSpecialCmd(DaliSpecialCmd::SET_DTR, r);
        sendSpecialCmd(DaliSpecialCmd::SET_DTR1, g);
        sendSpecialCmd(DaliSpecialCmd::SET_DTR2, b);
        sendSpecialCmd(DaliSpecialCmd::ENABLE_DT, 8);
        sendCmd(DaliCmdExtendedDT8::SET_TEMP_RGB_LEVEL);
        sendSpecialCmd(DaliSpecialCmd::ENABLE_DT, 8);
        sendCmd(DaliCmdExtendedDT8::ACTIVATE);
```