hubsif/arduino-dali #14: is it possible to use your library with stm32?

gonzzzales, 2024-12-04:


gonzzzales, 2024-12-04:
I tryed to change it for using with stm32, but unfortunetly it not work
I made this changes in DaliBus.cpp
`void DaliBusClass::begin(byte tx_pin, byte rx_pin, bool active_low) {
  txPin = tx_pin;
  rxPin = rx_pin;
  activeLow = active_low;

  // init bus state
  busState = IDLE;

  // TX pin setup
  pinMode(txPin, OUTPUT);
  setBusLevel(HIGH);

  // RX pin setup
  pinMode(rxPin, INPUT);
  //---attachPinChangeInterrupt(digitalPinToPinChangeInterrupt(rxPin), DaliBus_wrapper_pinchangeISR, CHANGE);
  attachInterrupt(digitalPinToInterrupt(rxPin), DaliBus_wrapper_pinchangeISR, CHANGE);

  // Timer setup
  //---Timer1.initialize(DALI_TE); // set timer to time of one half-bit
  //---Timer1.attachInterrupt(DaliBus_wrapper_timerISR);
  Timer1.pause();                                             // останавливаем таймер перед настройкой
  Timer1.setPeriod(DALI_TE);                                       // время в микросекундах
  Timer1.attachInterrupt(TIMER_UPDATE_INTERRUPT, DaliBus_wrapper_timerISR);  // активируем прерывание
  Timer1.refresh();  
  Timer1.resume();
}`

hubsif, 2024-12-04:
Hi @gonzzzales!
I don't have a STM32 board or any experience with it.
I don't think either the TimerOne and PinChangeInterrupt libraries used in this library support STM32. But you can have a look at @thewhobox's pull request #13 (which I haven't found the time to review yet). It uses a generic timer library instead that also supports STM32. However, you'd still have to configure and test it for STM32 yourself.

thewhobox, 2024-12-04:
Hey guys,

The Timerinterrupt_Generic Supports STM32.
Ist would need small changes to support it but thats easy.
I can add These changes to the pull Requests in the next days.

kubukoz, 2024-12-05:
Hi @hubsif, any chance you could share more about how this would be set up with an arduino, from scratch? The wiring in particular is a complete mystery to me.

gonzzzales, 2024-12-05:
@thewhobox i tryed to do it with rogerclarc core. In other projects Timer1 and pinchangeinterrupts work fine. My changes you can see above. Unfortunetly blink example didn't want to work))

thewhobox, 2024-12-09:
I added support for stm32 in my pr #13 (or this [branch](https://github.com/thewhobox/arduino-dali/tree/pr)).
Due to i dont own a stm board its not tested, but it compiles without errors.

thewhobox, 2024-12-09:
@kubukoz you need a pcb to receive or send dali commands.
You can find circuits online ([here ](https://electronics.stackexchange.com/questions/438615/need-help-understanding-dali-circuit-and-220-vac-protection)for example) or buy a finished product (dali click, led-warrior or waveshare dali2).

Then its only conencting RX and TX from pcb to your microcontroller.

Martsola, 2025-01-11:
I have also done some testing with this library and also with the thewhobox fork and I would need some help. My goal is to make bus powered remotely configured DALI control device which is using less than 2mA from DALI bus. I have a working send-only capable prototype using Arduino pro mini (Atmega328 8MHz@3.3V), buck regulator, diode bridge and transistor which uses 1,5mA without any software optimizations so I think my goal should be realistic.

For code testing I have used Waveshare Dali2 board (https://www.waveshare.com/pico-dali2.htm) and Arduini compatible boards listed below.

**Here are my current results with different hw platforms and libraries:**

### Atmega328 @16MHz/8MHz (Arduino Pro Mini 5V & 3.3V versions):
**Hubsif Dali library:**
Compiles OK, can send Dali commands if receive pin is not connected. If receive pin is connected, then then send is not sending anything.

**Thewhobox fork:**
Compiles almost by adding these lines to DaliBus.h:	
```
#define DALI_TIMER 2
#define USE_TIMER_2 true
```
  
Compile fails with error:
```
(symbol from plugin): In function `__vector_7':(.text+0x0): multiple definition of `__vector_7'
(symbol from plugin): In function `ITimer2':(.text+0x0): multiple definition of `ITimer2'
```

### STM32F103C8(Bluepill):
**Thewhobox fork:**
Compiles ok, but code fails. Nothing happens.
Even only setting the built in led ON fails.
If I comment "#include <Dali.h>" then rest of the non DALI related code starts working.

### STM32G031F8P6:
**Thewhobox fork:**
Compiles ok, but I do not yet have this board available to test. This is my low power alternative for Atmega.

### ESP32 S2 (Wemos/Lolin S2 Mini):
**Thewhobox fork + Arduino core for the ESP32 2.0.17:**
(Using esp32 core 3.x will fail to compile due timer library incompatibility)
With esp32 core 2.0.17 code compiles ok.
   
I also needed to add: 
```
#define DALI_NO_COLLISSION_CHECK
#define DALI_TIMER 1
```
By defining DALI_TIMER 1 I get compile succeed. Send works but without defining DALI_NO_COLLISSION_CHECK the sent data is corrupted. Corruption happens only if receive pin is connected. With DALI_NO_COLLISSION_CHECK defined or receive pin disconnected the sending works.
Also Dali activity detection works but Dali receive function is never triggered.

thewhobox, 2025-01-17:
You could add some println to see where it crashes.
I suggest its in the begin method of the DaliClass.
There is all the code to start timer and initiate the pin interrupt.

thewhobox, 2025-09-08:
It should work with rp2040 and esp.