hubsif/arduino-dali #16: Compaile error with Arduino Pro Mini and Thewhobox fork

Martsola, 2025-01-12:
I can get **Thewhobox fork** to compile almost by adding these lines to DaliBus.h:	
```
#define DALI_TIMER 2
#define USE_TIMER_2 true
```
But compile still fails in the very end of the process:
```
C:\arduino\sketches\12EF855E9E518841BF146FCF221D7C58\libraries\Dali\Dali.cpp.o (symbol from plugin): In function `__vector_7':
(.text+0x0): multiple definition of `__vector_7'
C:\arduino\sketches\12EF855E9E518841BF146FCF221D7C58\sketch\Dali.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\arduino\sketches\12EF855E9E518841BF146FCF221D7C58\libraries\Dali\Dali.cpp.o (symbol from plugin): In function `__vector_7':
(.text+0x0): multiple definition of `ITimer2'
C:\arduino\sketches\12EF855E9E518841BF146FCF221D7C58\sketch\Dali.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\arduino\sketches\12EF855E9E518841BF146FCF221D7C58\libraries\Dali\DaliBus.cpp.o (symbol from plugin): In function `ITimer2':
(.text+0x0): multiple definition of `ITimer2'
C:\arduino\sketches\12EF855E9E518841BF146FCF221D7C58\sketch\Dali.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\arduino\sketches\12EF855E9E518841BF146FCF221D7C58\libraries\Dali\DaliBus.cpp.o (symbol from plugin): In function `ITimer2':
(.text+0x0): multiple definition of `__vector_7'
C:\arduino\sketches\12EF855E9E518841BF146FCF221D7C58\sketch\Dali.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
```
Any ideas what is causing this?

Martsola, 2025-01-14:
I moved the line
`#define USE_TIMER_2 true`
to the beginning of DaliBus.cpp and I got rid of those linker errors.

Now I have same functionality like with Hubsif library:
-Sending DALI frames works
-Sending fails if receive pin is connected (nothing is sent to the bus)
-Receive callback is never triggered
-Receive activity callback is never triggered

Martsola, 2025-01-15:
Waveshare board has inverted tTX and not inverted RX.
As temporary fix I changed

```
  #elif defined(ARDUINO_ARCH_AVR) || defined(ARDUINO_ARCH_STM32)
  #define getBusLevel (activeLow ? !digitalRead(rxPin) : digitalRead(rxPin))
```
  
  to:
  
```
  #elif defined(ARDUINO_ARCH_AVR) || defined(ARDUINO_ARCH_STM32)
  #define getBusLevel (activeLow ? digitalRead(rxPin) : digitalRead(rxPin))				//inverted for waveshare board
```

Now sending works reliably also if receive pin is connected.

I still have issues that:
-Receive callback is never triggered
-Receive activity callback is never triggered

Same code without changes above works with ESP32 S2 + Waveshare Dali2 interface

Martsola, 2025-01-17:
Ok, the last one was pretty stupid bug. At some point I had moved TX and RX from pin 2 and 3 to something else. In 328 those are only ports usable for external interrupts.

I also tested that I can also use timer 1 same way like timer 2 described above.

closed 2025-01-17 by Martsola

hubsif, 2025-01-21:
Hi @Martsola!
Sorry for not having responded earlier, I was too busy.
I'm glad you were able to sort out all your issues, though!